t-edson/PicPas #1: Synedit

Alexey-T, 2016-11-27:
Hi again. You could make the PicPas on ATsynedit, and then make better lexer for pascal for pic (it is easy to make lexer, no code)

t-edson, 2016-11-28:
Hi Alex.
I didn't know, ATsynedit had a lexer. Where can I find information?

Alexey-T, 2016-11-28:
You can see CudaText. it uses ATsynedit with 20 preinstalled lexers. I can make lexer for pic pascal

t-edson, 2016-11-28:
It would be interesting see a lexer on ATsynedit for PicPas. But the idea of the design in PicPas is to be simple. Simple, because use libraries in layers:  SynFaci

t-edson, 2016-11-28:
It will be necessary to exist some equivalent to t-Xpres for ATsynedit. It could be developed, of course, but I'm not so familiar with ATsynedit. Maybe later, I will port some code to ATsynedit. I think I would start using it, in some project who need an editor, an editor with the features that ATsynedit can offer and SynEdit not.

Alexey-T, 2016-11-28:
I see, okay. Later create new issue (GH) and post 2-3 example fls (of PicPas), so i create hiliter

t-edson, 2016-11-28:
Good. I will create a new document defining the language of PicPas. It's going to change to follow the Modula-2 style.

Alexey-T, 2016-11-28:
Not document- just example files, i use examples

Alexey-T, 2016-11-28:
PicPas is only in your program, other apps use it too?

t-edson, 2016-11-28:
Well, I will prepare some examples. PicPas is still in development.  I don't know about other programs using PicPas.  I know about some people using SynFacilSyn, SynFacilCompletion, and even t-Xpress, but not about PicPas.

closed 2017-05-23 by t-edson