t-edson/PicPas #4: Error en traspaso de parámetros tipo word a procedimientos

AguHDz, 2017-05-22:
Cuando se intenta traspasar un valor menor de 255 a un procedimiento con una variable de entrada tipo word el compilador comunica un error de tipo. Ejemplo:

```
procedure delay_us(microsegundos: word);
begin
  // cuerpo de procedimiento.
end;

begin                          
  delay_us(100);
end.

```
El compilador advierte de error "Type paremeters error" y no compila.
Si en lugar de 100 se pone 256 compila perfectamente. Por lo tanto, no admite traspaso de variables tipo word menores de 256 en la versión 0.6.4.

t-edson, 2017-05-22:
Lo que sucede, es que PicPas interpreta los literales numéricos, asignándole el menor tipo posible. Si le escribes "delay_us(100)", define a "100", como un tipo byte y como es estricto en los tipos, no permite pasar un byte a un word.  Hay una función del sistema llamada word(), que permite convertir un byte a word, pero aún no está implementada. Lo voy a revisar, para ver si sale en la 0.6.5.

t-edson, 2017-05-22:
Lo que podrías hacer ahora es crear dos procedimientos delay_us(), (que es como se maneja actualmente delay_ms), uno con parámetro byte y otro con parámetro word. PicPas soporta la sobrecarga de procedimientos. Pero el soporte al tipo word es limitado por ahora.

AguHDz, 2017-05-22:
"PicPas soporta la sobrecarga de procedimientos."... Interesante, no lo sabía.

AguHDz, 2017-05-22:
Efectivamente, soporta sobrecarga de procedimientos, lo que me parece muy útil a la hora de crear librerías. Este código compila perfectamente:
```
procedure delay_us(microsegundos: word);
begin
  // cuerpo de procedimiento.
end;

procedure delay_us(microsegundos: byte);
begin
  // cuerpo de procedimiento.
end;

begin                          
  delay_us(100);
  delay_us(1000);
end.

```

closed 2017-05-22 by AguHDz