t-edson/PicPas #13: Directiva include en uses

AguHDz, 2017-08-24:
Parece de la directiva DEFINE tiene problemas cuando se relaciona con uses:

No compila:
```
// AQUI SE DEFINE LA FUNCION USADA PARA IMPRIMIR LOS NUMEROS EN UN DISPOSITIVO DE SALIDA.
{$DEFINE SALIDA_LCD}
{$DEFINE FUNCION_PRINT_DIGITO=LCD_WriteChar}  // Libreria LCDLib_4bits
{$DEFINE LIBRERIA_PRINT_DIGITO=LCDLib_4bits}
//{$DEFINE SALIDA_UARTSOFT}
//{$DEFINE FUNCION_PRINT_DIGITO=UARTSoft_SendChar}  // Libreria UARTSoftLib_8MHz_1200bps
//{$DEFINE LIBRERIA_PRINT_DIGITO=UARTSoftLib_8MHz_1200bps}

uses PIC16F877A, Math, {$LIBRERIA_PRINT_DIGITO};
```
Da error en línea de 2º uses.

Esto tampoco compila:
```
// AQUI SE DEFINE LA FUNCION USADA PARA IMPRIMIR LOS NUMEROS EN UN DISPOSITIVO DE SALIDA.
{$DEFINE SALIDA_LCD}
{$DEFINE FUNCION_PRINT_DIGITO=LCD_WriteChar}  // Libreria LCDLib_4bits
//{$DEFINE SALIDA_UARTSOFT}
//{$DEFINE FUNCION_PRINT_DIGITO=UARTSoft_SendChar}  // Libreria UARTSoftLib_8MHz_1200bps

{$IFDEF SALIDA_LCD}
  uses PIC16F877A, Math, LCDLib_4bits;
{$ENDIF}
{$DEFINE SALIDA_UARTSOFT}
  uses PIC16F877A, Math, UARTSoftLib_8MHz_1200bps;
{$ENDIF}
```
En linea uses, mensaje: Error Indentificador esparado.

t-edson, 2017-08-25:
Hola Agustín. El primer error, es simple de corregir. Es un tema de habilitar el procesamiento de directivas en ese punto. Estará corregido para la versión 0.7.5.
El segundo error que mencionas, no lo veo como error, proque estás tratando de incluir dos sentencias USES. Imagino que has querido poner {$IFDEF SALIDA_UARTSOFT}

AguHDz, 2017-08-26:
Tienes razón , el fallo del 2º error es mío.
```
{$IFDEF SALIDA_LCD}
  uses PIC16F877A, Math, LCDLib_4bits;
{$ENDIF}
{$IFDEF SALIDA_UARTSOFT}
  uses PIC16F877A, Math, UARTSoftLib_8MHz_1200bps;
{$ENDIF}
```
en la versión 0.7.4 que lo acabo de probar funciona perfectamente.

t-edson, 2017-08-26:
Quería comentar, adicionalmente, que las directivas están aún en modo experimental. Es posible que la sintaxis cambien, porque he encontardo ciertos conflictos con el formato hezadecimal de números, que en Pascal son de la forma: $FFFF al igual que los identifiacdores de las macros $nombre. En todo caso, los cambios serán sencillos. Y se estará indicando, en al versión en la que aparezca.

AguHDz, 2017-08-28:
Probado en versión 0.7.5.
```
// AQUI SE DEFINE LA FUNCION USADA PARA IMPRIMIR LOS NUMEROS EN UN DISPOSITIVO DE SALIDA.
{$DEFINE SALIDA_LCD}
{$DEFINE FUNCION_PRINT_DIGITO = LCD_WriteChar}  // Libreria LCDLib_4bits
{$DEFINE LIBRERIA_TERMINAL_SALIDA = LCDLib_4bits}
//{$DEFINE SALIDA_UARTSOFT}
//{$DEFINE FUNCION_PRINT_DIGITO = UARTSoft_SendChar}  // Libreria UARTSoftLib_8MHz_1200bps
//{$DEFINE LIBRERIA_TERMINAL_SALIDA = UARTSoftLib_8MHz_1200bps}

uses PIC16F877A, Math, {$LIBRERIA_TERMINAL_SALIDA};
```
Ahora funciona perfectamente.

closed 2017-08-28 by AguHDz