t-edson/PicPas #14: Reutilización de variables locales (al menos si tienen el mismo nombre)

AguHDz, 2017-08-25:
PicPas tiene un problema con el uso de memoria RAM cuando se amplean muchas variables globales. De hecho en este momento es más interesante usar variables globales para enviar datos a las funciones.

Creo que solucionar este problema de manera general no sea sencillo, o habrá que solucionarlo usando variables auxiliares reutilizables, pero creo que sería sencillo que distintas funciones usan un mismo nombre de variables de entrada y del mismo tipo, las pudieran reutilizar.

Por ejemplo, tengo estas cinco funciones:
```
Print_Digito (numero : byte);
procedure Print_Numero_8bit(numero : byte);
procedure Print_Numero_16bit(dato : char);
procedure Print_Numero_32bit(dato : char);
procedure Print_Numero_64bit(dato : char);
```
Se compila:
```
numero EQU 0x038
numero EQU 0x039
dato EQU 0x03A
dato EQU 0x03B
dato EQU 0x03C
```
Ya que hay variables locales que se llaman igual y son del mismo tipo, tal vez resultara sencillo que PicPas lo compilara como:
```
numero EQU 0x038
dato EQU 0x039
```
Aunque el problema vendría en caso de llamadas entre funciones compartan el mismo nombre y tipo de variable de entrada. En ese caso si que sería necesario crear alguna variable auxiliar antes de llamadas entre este tipo de funciones.

Cuando se empiezan ha hacer librerías un poco complicadas, las variables locales se comen literalmente la memoria RAM del microcontrolador, por lo que hay que acabar usando el método del lenguaje ensamblador de crear variables de intercambio globales y llamar a las funciones como si fueran procedimientos.

t-edson, 2017-08-25:
La reutilización de variables locales, es un tema pendiente aún en PicPas. La implementación eficiente no es sencilla, proque requeire analizar el árbol de llamadas, entre procedimientos. Inicialmente pensé en que las variables globales en ensamblador aparezcan como <nombre_proced.>_<nombre_variable_local>, para que se puedan diferenciar dentro del código ensamblador, pero al momento de reutilizar las variables, estyo ya no sería posible. Es por eso que actualmente usan el nombre local, y aparecen en ensamblador, con el mismo nombre. Voy a revisarlo, pero no sé cuándo de podrá implementar. Imagino que una implementación sencilla, se puede hacer pronto.

AguHDz, 2017-08-26:
Como está ahora, lo cierto es que se entiende muy bien el código ensamblador generado y ponerle a las variables locales el nombre <nombre_proced.>_<nombre_variable_local> hubiera sido un acierto para la mejor comprensión del código generado, pero eso se puede suplir con los comentarios que se generan en ensamblador, se tome la decisión que se tome, que creo que al final será la de usar variables auxiliares, sería muy recomendable poner a modo de comentario que variable local contiene la auxiliar. Dentro de mi ignorancia en la materia, me parece que lo del comentario en el código ensamblador generado sería fácil de implementar.

t-edson, 2018-01-05:
La versión 0.8.2 utiliza un algoritmo mejorado para optimización de variables locales. Así que ahora se reduce el consumo de RAM. El nombre que se usa sigue siendo el mismo nombre de la variabel local en Pascal y eso puede dar lugar a confusión. El problema del nombre de la variable en el ensamblador es otro tema y esta ligado a obtener información de bancos al momento de desensamblar, algo que es un poco complicado, porque PicPas hace el desensamblado se hace desde el código binario, ya que el compilador no genera ASM, sino binario directamente.

closed 2018-01-08 by t-edson