t-edson/PicPas #15: Herencia de parámetros del programa principal en el uso de Librerías

AguHDz, 2017-08-29:
Para hacer librerías genéricas para PicPas, nos encontramos con la dificultada de que es no es posible definir parámetros o variables en la librería que se deban declarar en el programa que la vaya a usar.

Es un problema importante, ya que por ejemplo, para poder usar una librería es necesario definir en la propia librería el microcontrolador y cargar su librería de registro SFR (o escribir en el propio programa la posición de memoria de cada registro SFR que utilice). Así es muy complicado hacer una librería genérica. Tampoco hereda o reconoce directivas DEFINE del programa principal. Todo ello imposibilita la realización práctica de librerías genéricas, tan solo se pueden hacer para un o unos determinados microcontroladores y siempre que coincidan las posiciones de los registros SFR, pero tampoco es correcto ya que el dato de memoria RAM y Flash de un micro a otro también varía, por lo que en la práctica casi sería necesario hacer una versión diferente de cada librería para cada microcontrolador PIC soportado.

En un principio, con que pudiera heredar los uses y define del programa que usa la librería sería suficiente, o al menos definir algún tipo de identificador tipo EXTERNAL para variables y constantes.

t-edson, 2017-08-29:
Hola Agustín, Los DEFINE del programa principal, si los debe reconocer, siempre y cuando estén declarados, antes del USES.  Si defines esto:
```
{$PROCESSOR PIC16F877}
{$FREQUENCY 4Mhz}
{$DEFINE aaa}
uses PIC16F84A, MiUnidad;
...
```
El símbolo "aaa", es visible desde dentro de PIC16F84A y MiUnidad. También el modelo del PIC es accesible en estas unidades, porque se ha definido {$PROCESSOR PIC16F877}, previamente. 
Lo que está pendiente es, alguna forma de identificar al modelo, usando directivas. Pero más importante aún, sería acceder a las características del hardware del procesador usado. Algo así como:
{$IF HAS_TIMER0}
....
{$ENDIF}
o
{IF HAS_PORTD}
...
{$ENDIF}

t-edson, 2017-08-29:
Ahora, lo que se define en una unidad, como PIC16F84A, son realmente nombres para bits o bytes de la memoria del dispositivo, no las características del hardware. Eso se define cuando se hace {$PROCESSOR PIC16F877}. Por eso es importante que se haga siempre al inicio. 
Acceder por nombres a los registros, desde dentro de una librería, se podría hacer con una suerte de:
```
{$IF model='PIC16F877A'}
   uses PIC16F877A;
{$ENDIF}
{$IF model='PIC16F877A'}
   uses PIC16F84A;
{$ENDIF}
...
```
Pero es un código laborioso, por eso, decía que era mejor,acceder a características del hardware, para implementar librerías, en lugar de a los modelos de los dispositivos.

AguHDz, 2017-08-29:
Por qué no creas un tipo de fichero de texto, que no tiene por qué ser compatible con código Pascal donde se puedan de definir de manea accesible para cualquiera todos los parámetros de cada procesador, así cuando se coloca la etiqueta {$PROCESSOR PIC16F877} automáticamente el compilador interpreta todos los datos de ese fichero (nombres y posición de registros SFR, tamaños de memoria, fuses o lo que se vaya necesitando), sería una especie de fichero de configuración de cada microcontralador sin que hiciera falta añadirlo en uses o los nombres de los fuses, etc. Mira por ejemplo cómo lo hace Proton IDE, tiene un fichero para cada microcontrolador en el que define absolutamente todos sus parámetros y a ese fichero se le puede dar cualquier formato... creo que la solución debería ir por ese camino, si se quiere hacer una cosa estándar y utilizable por cualquier nuevo usuario o trabajar en grupo creando librerías.

Y que los fichero uses de las librerías esperen el tipo de {$PROCESSOR PICxxxx} del programa que las use, pero que no de errores de variables no definidas cuando por ejemplo se usar registros SFR.

AguHDz, 2017-08-29:
Lo de los DEFINE antes de uses ya es un avance importante. ¿se podía usar la directiva INCLUDE en dentro de librerías? Si es así no importa lo complicado que sea codificar un método, solo sería necesario escribirlo una vez y lo incluirían todas las librería.

Creo que herramientas casi que hay para hacer muchas cosas, lo que hay es que empezar a darle uso práctico de manera que salgan a relucir la deficiencias que posiblemente se puedan solucionar sin mucha complicación. Tal vez no seamos conscientes, pero PicPas ha dado pasos de gigante durante el último mes, tanto que hay cosas que ni he podido probar o que están no documentas en el manual de usuario.

AguHDz, 2017-08-29:
Mañana intentaré tomar la librería de escritura en displays LCD, que tengo ya muy probada, pero que debo modificar cada vez que uso un modelo distinto de PIC, a ver de alguna manera consigo que usando DEFINEs se pueda utilizar para cualquier microcontrolador de los contemplador por PicPas. Y con el resultado práctico, se pueden ir sacando conclusiones. Es posible que se pueda, pero que no sepa como hacerlo todavía.

t-edson, 2017-08-29:
Precisamente, la idea de usar {$PROCESSOR PIC16F877}, era que sería una forma resumida de hacer algo como:

```
{$DEFINE MAX_FREQ = 20000000}
{$DEFINE NPINS = 40}
{$DEFINE NUMBANKS = 4}
{$DEFINE NUMPAGES = 4}
{$DEFINE GPRSTART = $20}
{$SET_STATE_RAM 0x00-0x1F = IMP_SFR}
{$SET_STATE_RAM 0x83 MAPPED TO 0x03}
{$DEFINE HAS_TIMER0=TRUE}
...
```

Aunque no está soportado, aún. Todo esto se podria definir en el mismo archivo fuente o en un archivo adicional, y lugeo accederlo con {$PROCESSOR PIC16F877} o con {$INCLUDE PIC16F877.def}

Lo que no se puede incluir aquí (no se corresponde), es los nombres Pascal que se usan para los registros, como STATUS_Z o algo así. Esos nombres son propios de la librería y en la sintaxis de Pascal.

t-edson, 2017-08-29:
Lo que si se podría hacer, es usar una directiva para incluir psoteriormente una unidad en el USES del programa. Algo así, como: {$USES PIC16F877.pas}, para que se accedan a los nombres de lso registros. Para

t-edson, 2017-08-29:
Lo que si se podría hacer, es usar una directiva para incluir psoteriormente una unidad en el USES del programa. Algo así, como: {$USES PIC16F877.pas}, para que se accedan a los nombres de los registros. 
Lo ideal sería, como dices poner toda la descripción del hardware y registros en un solo archivo. Eso se podría lograr tal vez, si se incuyen todas las directivas de definición de hardware, dentro de la misma unidad-librería. Habría que hacer pruebas. Voy a activar el acceso a algunas propiedades del hardware para hacer algunas pruebas, en esta versión, a ver que sorpresas aparecen en el camino. De momento ya encontré unos problemas en las directivas, que tengo que solucionar.

AguHDz, 2017-08-30:
Excelente. Como te decía, creo que la herramientas están, pero aún no se como usarlas. Mira la solución para el problema del modelo de microcontrolador en las librerías. Este código compila sin errores:
EN PROGRAMA:
```
{$PROCESSOR PIC16F84A}
{$FREQUENCY 8Mhz}
{$DEFINE MODEL_PIC=PIC16F84A}

program LCD16x2_4B_Lib_DEMO;

uses {$MODEL_PIC}, LCDLib_4bits_NEW; 
```
EN LIBRERIA:
```
unit LCDLib_4bits_NEW;

interface

{$IFNDEF MODEL_PIC}
  {$MSGBOX 'Se debe indicar el modelo de microcontrolador al inicio del programa. (Por ejemplo: $DEFINE MODEL_PIC=PIC16F877A)}
{$ENDIF}

uses {$MODEL_PIC}, LCDLib_Const;
```
Importante avance. No solo compila,... El programa funciona!

AguHDz, 2017-08-30:
Sin embargo, para definir los pines me encuentro con el problema de que la directiva DEFINE no se reconoce todavía en el área de definición de variables:
```
{Description of the program.}
program nombre;
{$DEFINE POSICION=$0030}

var
  VERIFICACION : byte absolute $0030;
  PRUEBA       : byte absolute {$POSICION};

begin
  // nada que hacer.
end.
```
ERROR  {7,28]: Error: Se esparaba dirección numérica.

t-edson, 2017-08-31:
Existe ya, una variable interna para acceder al modelo del PIC, pero aún no es accesible. Dejame ver si la puedo activar. 
Lo de la macro, debe ser fácil de corregir. Ahora lo reviso.

AguHDz, 2017-08-31:
Una vez esté accesible el modelo de microcontrolador y el reconocimiento de macros en el área de variables, se pueden empezar a hacer librerías más en serio. Solo sería cuestión de que las organice un poco, pero tendrá librerías:
- Displays LCD (aunque para un uso más a lo que estamos acostumbrado deberíamos definir datos tipo cadenas String en PicPas. En principio, podrían ser sólo cadenas de textos como constantes, implementar su uso y manejo. Y dejar para más adelante las cadenas de texto o Arrays en general como variables).
- Comunicación Serie mediante Software.
- Lectura/Escritura de memoria EEPROM interna del microcontrolador.
- WatchDog del microcontrolador.
- Comunicación I2C mediante software. Casi hecho, pendiente solo de probar.
- Contador de tiempo usando Timers 0 o 1.
- Funciones matemáticas aún no implementas en PicPas.
- Uso de convertidores A/D del micro. Pendiente de probar.

Y me sería muy sencillo por tenerlas ya para micros 8051:
- Uso de integrados multiplexores externos para puertos.
- Manejo de displays led 8x8 y de 7 segmentos usando el integrado MAX7219.
- Control de cientos de leds usando comunicación SPI con integrado MAX7219.
- Uso del registro de desplazamiento 74595 para extender los pines de salida.
... y seguro que tengo por ahí más cosas hechas que puedo adaptar en minutos.

t-edson, 2017-09-01:
Ya está arreglado el problema del reconocimiento de directivas en la declaración  de variables.  Puedes descargar de GitHub, la nueva version 0.7.6, que aún no está cerrada. También se ha activado el acceso a diversas variables del sistema. El modelo del PIC, lo puedes acceder a través de la variable {$PIC_MODEL}, y hay diversas variables más. Otras más se irán activando posteriormente.

t-edson, 2017-09-01:
Las cadenás en PicPas, aún están pendientes. Como en otros casos, el diseño está en papel, pero falta la implementación. 
De acuerdo a lo planificado, en la versión 0.7.7 se debe incluir definiciones básicas de tipos, definidos por el usuario. La versión 0.7.8 debe incluir manejo de arreglos sencillos, y en la versión 0.7.9, deben aparecer las cadenas. Así en la versión 0.8, debemos estar ya trabajando con cadenas.

AguHDz, 2017-09-01:
Ok, pues vamos a probar. La librería LCD está prácticamente hecha. Una misma librería es válida para cualquier microcontrolador y el modo de bus de datos de 4 y 8 bits.

Por cierto, como comentamos en otro lugar, esta vez me he fijado bien en fallos a la hora de recuperar el tema (colores, tipo texto, etc.). No he descargado todos los ficheros, solo el nuevo ejecutable PicPas-win32.exe, y aparentemente se ha solucionado el problema y recupera todos los colores. Lo único que he tenido que modificar manualmente ha sido el tipo de letra, eso parece que falla todavía.

AguHDz, 2017-09-01:
Perfecto. PicPas ya tiene una librería, creo que decente, para el manejo de displays LCD (https://github.com/AguHDz/PicPas-Library/tree/master/LCD), aunque se irán añadiendo nuevos tipos de conexiones y cuando esté disponible, la impresión de cadenas de textos y números.

Para mantener más organizado el tema de la librerías, he creado un nuevo repositorio donde las iré añadiendo: https://github.com/AguHDz/PicPas-Library , en el que mantendré siempre las últimas versiones de librerías que haga yo o cualquiera que se quiera unir al proyecto. Eres libre de añadirlas a cada nueva versión de PicPas. Pero sería recomendable que no las modifiques o cambies sus nombres, ya que todo el conjunto mantendrá dependencias que me encargaré de actualizar con cada nueva versión de librería. Ejemplos de uso puedes añadir tantos como quieras.

AguHDz, 2017-09-01:
Queda un tema por resolver. Ahora esto no funciona:
```
{$IFNDEF PIC_MODEL}
  {$MSGBOX 'Se debe indicar el modelo de microcontrolador al inicio del programa. (Por ejemplo: $PROCESSOR PIC16F84A)}
{$ENDIF}
```
Aunque PIC_MODEL esté declarado a través de $PROCESSOR PIC16F84A siempre aparece la ventana de advertencia. De momento, se un tema menor, pero si se sigue avanzado pronto será necesario poder reconocer el valor de PIC_MODEL y que la librería vaya adaptando el código generado a las particularidad de cada microcontrolador (algo así supongo que tienes previsto:
```
{$IF PIC_MODEL = PIC16f84A}
......
{$ENDIF}
```
o
```
{$IF BAUDIOS = 9600}
.......
{$ELSIF BAUDIOS = 4800}
.......
{$ENDIF}
```

t-edson, 2017-09-01:
Ya está corregido, la verificación de PIC_MODEL. Es que ahora se manejan macros y variables dentro de las direvtivas y hay que ampliar algunas instrucciones a ambos tipos de elementos. El manejo de las directivas es ahora más complejo (Es todo un capítulo en el Manual de Usuario). Hay otro lenguaje dentro de las directivas, con varaibles, macros, operadores, expresiones, funciones, y condicionales. Lo bueno es que es un lenguaje interpretado y puede ampliarse fácilmente. Por ejemplo ahora se tienen algunas pocas funciones definidas, pero se pueden ir añadiendo más, conforme se vea que se requiera. La idea es que se pueda usar el lenguaje de las directivas para manipular cadenas, y comparar valores o crear tablas. Por ejemplo se podria crear librerías de retardos, para toda frecuencia de reloj. Sería algo así como:
{$IF PIC_FREQUEN=1000000}
...
{$ELSIF PIC_FREQUEN=2000000}  //no implementado aún
...
{$ELSIF PIC_FREQUEN=4000000}  
...
{$ELSIF PIC_FREQUEN>8000000}  
...
{$ELSE}
...
{$ENDIF}

t-edson, 2017-09-01:
Con respecto a las librerías, me parece bien que centralices el archivo. Ya es hora de que PicPas tenga algunas librerías estándar. Voy a ir moviendo librerías al repositorio de PicPas, conforme vaya revisándolas. De momento me parece que se debería incluir también, los bits de configuración dentro de cada librería para PIC. 
Un detalle adicional, al momento de crear las librerías, sería considerar que en el futuro, se espera agregar nuevos dispositivos, definiéndolos por directivas u otro medio alternativo (Actualmente se puede moficar parámetros como PIC_MODEL o PIC_NUMBANKS). Así , que tal vez sería mejor, más que ver el modelo del PIC, acceder a características comunes del hardware, del Pic como:

{$IF CORE='14BITS' AND PIC_HAS_PORTD AND PIC_GPR_START=0x20}
...
PicPas almacena información detallada del dispositivo, que es posible acceder desde las directivas. Esto es porque, PicPas tiene un modelo del hardware (por ahora solo con núcleo de 14 bits), que le permite realizar simulación de código, al estilo de MPLab. Esta funcionalidad ya está avanzada y pronto aparecerá como una funcionalidad adicional de la IDE, aunque algo muy básico al principio.

Con respecto a lo de los temas, yo veo que si guarda el tipo de letra. Lo he probado al menos, en mi PC, modifiacando un tema existente.

AguHDz, 2017-09-01:
Acabo de bajar la versión 0.7.6 completa y creo que lo del tipo de letras fue por, como comentaba, haber bajado solo el ejecutable que pensé era una versión 0.7.5 corregida. Ahora si toma todos los colores y tipos de letra, pero para acabar la personalización he tenido que copiar el fichero PicPas-win32.xml de la anterior versión, por lo que he visto, entiendo que este es como la configuración del "espacio de trabajo personal" donde se guardan el tamaño de los iconos, las herramientas personalizadas, los últimos ficheros abiertos, etc. que no tiene que ver con los colores y tipos de letra. Uno sería el "tema" y el otro el "espacio de trabajo".

En cuanto a las directivas "avanzadas", serán de gran ayuda para pulir y optimizar las librerías. Conforme se vayan poniendo en uso, e importante, documentando en el manual, las iré incorporando a las nuevas versiones de las librerías. Mi idea es disponer de unas librerías básicas que funcionen aceptablemente para que cualquiera pueda usarla, y a mi me sirvan para poder hacer algunos proyectos prácticos con PicPas. Y después, mantenerlas las días con todas las novedades que se vayan incorporando al compilador, o con las correcciones y mejoras que se detecten durante el uso práctico de las librerías.

Dentro de esas opciones avanzadas, he estado probando si cabría la posibilidad de mostrar una ventana con la configuración seleccionada para la librería usada mientras se compila, usando los condicionales y la directiva MSBOX, pero no se si estará implementado el encadenamiento de cadenas, para por ejemplo mostrar algo como esto:

`{$MSGBOX 'MICROCONTROLADOR: ', PIC_MODEL}`

y si habría posibilidad de añadir dentro de la cadena, el salto de línea, por ejemplo, con un #10,#13", un LFCR o algo como en C 'cadena\n'.

t-edson, 2017-09-01:
Lo de la concatenación de cadenas, si está implementada. Es:
`{ $MSGBOX 'MICROCONTROLADOR: ' +  PIC_MODEL}`
Esta indicado en el Manual, pero no de manera detallada, y se pierde con tanta infromación. Hay que mejorar esa parte de la documentación. Es que el tema de las directivas es cada vez más amplio, como ya había previsto desde antes.

Lo de los saltos de línea, aún no, pero es fácil incluirlo. No ovlidar que las directivas $WARNING y $ERROR, permiten mostrar mensajes de advertencia y error en el panel de mensajes. Está pendiente la directiva $INFO, para mostrar mensajes informativos.

Si usas $MSGBOX, el compilador se detiene hasta que se pulse "Aceptar".

AguHDz, 2017-09-01:
¿Estás subiendo el manual actualizado?, el de GitHub se quedó estancado en la versión 0.7.3

t-edson, 2017-09-01:
Sï. Pero algo debo estar haciendo mal. Déjame actualizarlo.

AguHDz, 2017-09-01:
Me parecía un poco extraño, pero supuse que con tanta actualización lo ibas dejando para más adelante. Lo cierto, es que sin el manual es como si las nuevas características no existieran, o las vas encontrando por tus comentarios o por simple casualidad, pero cada vez se hace más difícil seguirlo.

Como creo que el asunto que dio origen a esta conversación "Herencia de parámetros del programa principal en el uso de Librerías" está plena y satisfactoriamente resuelto, con tu permiso, lo doy por cerrado.

closed 2017-09-01 by AguHDz

t-edson, 2017-09-01:
Era un problema de actualización. No sé por qué no guardaba los cambios. De todas formas, en la página principal del gitHub, hay un resumen del manual, en inglés. Y sí, estoy de acuerdo en que el tema está cerrado y yo voy también, cerrando la versión 0.7.6.