t-edson/PicPas #16: Error al intentar mostrar variable de texto asignada con DEFINE en mensajes de directiva MSGBOX

AguHDz, 2017-09-01:
En el siguiente código de prueba:
```
{$IF LCD_BUS_DATA_BITS = 8}         // Si Modo bus de datos 8 BITS.
  {$DEFINE LCD_PORT_DATA_8BIT = PORTB}  // Puerto de bus de datos de 8 BITS.
  {$MSGBOX 'MICROCONTROLADOR: ' + PIC_MODEL}
  {$MSGBOX 'BUS DE DATOS DE ' + LCD_BUS_DATA_BITS + ' BITS'}
  {$MSGBOX 'PUERTO USADO: ' + LCD_PORT_DATA_8BIT}
{$ENDIF}
```
Los dos primeras MSGBOX funcionan perfectamente, pero el tercero muestra el mensaje de error:

**Identificador desconocido: PORTB**

y detiene la compilación.

Aclarar que si elimino la linea del error y la compilación se completa, LCD_PORT_DATA_8BIT (= PORTB) es perfectamente aceptado en la librería donde lo usa y compila hasta el final sin problemas.

t-edson, 2017-09-01:
Hasta donde yo veo, no es un error. Es solo que en el nuevo intérprete de directivas. la expresión:
{$MSGBOX 'PUERTO USADO: ' + LCD_PORT_DATA_8BIT}
 Sería equivalente a:
{$MSGBOX 'PUERTO USADO: ' + PORTB} 
Y en esa expresión, se esperaría que PORTB, sea una variable (y de tipo cadena). Pero está definida como una macro.
El código funcionaría si se hace:
```
{$PROCESSOR PIC16F84}
{$DEFINE LCD_BUS_DATA_BITS=8}
{$IF LCD_BUS_DATA_BITS = 8}         // Si Modo bus de datos 8 BITS.
  {$DEFINE LCD_PORT_DATA_8BIT = 'PORTB'}  // Puerto de bus de datos de 8 BITS.
  {$MSGBOX 'MICROCONTROLADOR: ' + PIC_MODEL}
  {$MSGBOX 'BUS DE DATOS DE ' + LCD_BUS_DATA_BITS + ' BITS'}
  {$MSGBOX 'PUERTO USADO: ' + LCD_PORT_DATA_8BIT}
{$ENDIF}
```

t-edson, 2017-09-01:
Pero yo te recomendaría que uses mejor:   {$SET LCD_PORT_DATA_8BIT = 'PORTB'}, en lugar de {$DEFINE ...
Y trabajar así con variables. 
Las variables son más apropiadas para trabajar con expresiones, dentro del lenguaje de directivas. Las macros, son útiles, para acceder a ellas desde código Pascal, y solo en algunos casos, proque se pueden usar variables también. 
En general diría que las variables deberían preferirse en lugar de las macros, pero es un tema que habrá que ver con el uso, porque este lenguaje, es algo nuevo y aún no lo he usado mucho.

AguHDz, 2017-09-02:
Ok, después de haber podido leer la versión actualizada del manual, me podría haber autocontestado. Pero probando la diferencia en entre DEFINE y SET, te comento que ambas funcionan igual con MSGBOX, pero no cuando se va a usar el contenido de la variable. Así por ejemplo:
```
{$DEFINE LCD_PORT_DATA_8BIT = 'PORTB'}  // Puerto de bus de datos de 8 BITS.
{$MSGBOX 'PUERTO USADO: ' + LCD_PORT_DATA_8BIT}
var
  LCD_DATA     : byte absolute {$LCD_PORT_DATA_8BIT}; // Puerto de datos 8 bits.
```
muestra la ventana MSGBOX sin problema, pero no compila, dando un error de sintaxis en la línea de definición de la variable LCD_DATA. Supongo que debido a que DEFINE sustituye el texto con las comillas.

Mientras que 
```
{$SET LCD_PORT_DATA_8BIT = 'PORTB'}  // Puerto de bus de datos de 8 BITS.
{$MSGBOX 'PUERTO USADO: ' + LCD_PORT_DATA_8BIT}
var
  LCD_DATA     : byte absolute {$LCD_PORT_DATA_8BIT}; // Puerto de datos 8 bits.
```
funciona y compila perfectamente.

Aunque en principio como dices en el manual puedan parecer comandos con la misma utilidad, creo que cada uno tiene una función distinta, DEFINE es un sustituto de cadenas de texto, mientras que SET es un auténtico intérprete de datos que será muy útil gracias a su capacidad de calcular de manera automática determinados parámetros de funcionamiento en función de las variables de entorno que se introduzca en el código del programa, como la velocidad y el micro usado. Otro paso adelante muy acertado. Vendrá muy bien en la librería de comunicación serie.

t-edson, 2017-09-02:
Así es. Las variables se evalúan y tienen un valor. Las varaibles cadenas, tienen pensado soportar operadores de cadena (la concatenación, ya se soporta) y métodos  como UpCase(), substr(), length(), ... de forma que se pueda manipular completamente el contenido de las cadenas. 
Las macros son solo un reemplazo literal de texto. Y el lenguaje de las directivas es de alto nivel, y soporta expresiones.
Realmente lo que contiene una macro, es código (código fuente) que es reemplazado, tal cual es definido. Es como manejar un  nuevo tipo de dato, que se llamaría "código fuente". Algo parecido a la programación funcional.
Pero la potencia de las macros, está cuando se define en base a otras macros. En este caso, la expansión de macros, se hace de forma recursiva y puede crear contenido dinámico, de forma bastante elaborada.
Además, está planeado que las macros soporten parámetros, de modo que se tenga algo así, como: 
`{$DEFINE mayor(a,b) = IF a>b THEN a ELSE b;}`
Que sería el equivalente de una función INLINE.

AguHDz, 2017-09-02:
Si como dices es crear un lenguaje interpretado dentro del compilador. Recuerdo haber probado la directiva DEFINE en algún compilador de C para "programar en español" y resultó muy sencillos, solo había que darle el nombre adecuado en español al macro creado con DEFINE y los parámetros de entrada necesarios. Son temas de las que el común de los usuarios no se preocupa ni de conocer, sólo se tiene un conocimiento muy superficial de este tipo de herramientas tan importantes para la optimización del código generado..

PicPas dispondría en la práctica de tres "lenguajes": Pascal, ASM y el intérprete de directivas.

Por cierto, y solo a título de curiosidad porque en principio no tengo necesidad de usarlo: ¿Tienes previsto que en los bloque ASM se reconozcan las directivas de compilación?

t-edson, 2017-09-02:
Lo condideré, pero de momento no hay planes de incluir soporte para directivas dentro de blqoues ASM. 
* Primero, porque hay dificultades técnicas, debido a que los bloques ASM, son considerados como un solo token, y el lexer principal de Pascal, (el que procesa directivas) no identifica elementos individuales dentro de bloques ASM, sino que se usa otro lexer especial.
* Segundo, porque podrían usarse medios alternativos como constantes o variables Pascal, (controlados por directivas) que se pueden usar dentro de los bloques ASM.

t-edson, 2017-09-02:
Y como dices, ya PicPas tiene soporte para 3 lenguajes; Un ensamblador, un compilador y un intérprete. 
La particularidad del intérprete de directivas, es que es un lenguaje nuevo, e interpretado, así que se puede experimentar nuevos paradigmas, lejos de las limitaciones que impone el diseño de compiladores (que es más complicado y esta limitado por el hardware destino).

closed 2017-09-02 by AguHDz