t-edson/PicPas #17: Error grave en operación de suma de variables de tipo byte y word.

AguHDz, 2017-09-02:
Hay un error en la codificación de sumas cuando los dos operadores son variables.

Variables de tipo byte:
```
    ;bcd = bcd + decimal;
    0x02A movf decimal,w 
    0x02B addwf bcd,w 
    0x02C subwf bcd,w        <--------- ¿que hace aquí esta resta?
```
Y cuando las variables son tipo word:
```
    ;bcd = bcd + decimal;
    0x06F movf bcd@1,w 
    0x070 addwf decimal@1,w 
    0x071 movwf _H 
    0x072 movf bcd@0,w 
    0x073 addwf decimal@0,w 
    0x074 btfsc STATUS, 0 
    0x075 incf _H,f 
    0x076 movwf aux4    <----- a partir de aquí ¿? 
    0x077 movf bcd@1,w 
    0x078 subwf _H,w 
    0x079 btfss STATUS, 2 
    0x07A goto 0x07D 
    0x07B movf bcd@0,w 
    0x07C subwf aux4,w 
```

Son fragmentos de compilar estas dos funciones en una librería:
```
//***********************************************************************
//  FUNCION: DecToBCD
//  Devuelve el valor de entrada decimal en formato BCD de 2 digitos.
//***********************************************************************
procedure DecToBCD_2bytes(decimal : byte) : byte;
var
  bcd : byte;
begin
  if decimal>99 then exit($EE) end;  // Indica ERROR en valor decimal de entrada.
  bcd := 0;
  while decimal > 9 do
    bcd     := bcd + 16;
    decimal := decimal - 10; 
  end;
  bcd = bcd + decimal;
  exit(bcd);
end;

//***********************************************************************
//  FUNCION: DecToBCD
//  Devuelve el valor de entrada decimal en formato BCD de 4 digitos.
//***********************************************************************
procedure DecToBCD_4bytes(decimal : word) : word;
var
  bcd : word;
begin
  if ((decimal.high>99)
    OR ((decimal.high=99) AND (decimal.low>99)))
    then exit($EEEE) end;  // Indica ERROR en valor decimal de entrada.
  bcd := 0;
  while ((decimal.high>0) OR (decimal.low > 9)) do
    bcd     := bcd + word(16);
    decimal := decimal - word(10); 
  end;
  bcd = bcd + decimal;  
  exit(bcd);
end;
```
Por cierto, el operador += parece no estar implementado todavía aunque en el manual se diga que ya lo esta, o eso es lo que me pareció entender.
`bcd += decimal;`
Produce un error: Undefined operator: += for type: byte

t-edson, 2017-09-03:
Ahí veo que se está haciendo una comparación, en lugar de una asignación, por el operador "=". Debe ser por eso que se hace la resta. 
Estrictamente, en Pascal, no se debería permitir este tipo de expresiones aisladas (que no sean de asignación), pero en PicPas, lo he mantenido porque me es útil para depuración.
Ahora con respecto al operador +=, solo está implementado a modo experimental, para el tipo dword, que es en donde la optimización es más crítica. Pualtinamente lo iré implementando para los otros tipos.

AguHDz, 2017-09-03:
Pues es verdad. Error de código corregido y compila bien, según lo esperado. No es la primera vez que me ocurre eso de olvidar lo dos puntos y perder el tiempo buscando el fallo en algún sitio cuando lo tengo delante de las narices, jajaja...

Y en PicPas, además, con el añadido de que todavía hay que buscar posibles fallos en el compilador... Vaya pasatiempos que nos buscamos la "gente rara". ;)

closed 2017-09-04 by AguHDz