t-edson/PicPas #18: Decimales basura en cálculos con datos de variables de directiva SET

AguHDz, 2017-09-08:
Un pequeño detalle, probando el compilador este este código:
```
{
  Prueba de Directivas de compilador.
  PicPas 0.7.6
}

{$PROCESSOR PIC16F877A}

{$FREQUENCY 16MHz}
{$SET CICLO_MAQUINA = 4*1000000/PIC_FREQUEN}
{$MSGBOX 'CICLO MAQUINA = ' + CICLO_MAQUINA + ' us'}

{$SET dato1 = 100}
{$SET dato2 = 7}
{$SET RESULTADO = (DATO1\DATO2)}
{$MSGBOX 'RESULTADO: ' + RESULTADO}

program nombre;
var
  dato_bank0 : word absolute $020;  // Se localiza la variable en el Bank 0
  dato_bank1 : word absolute $0A0;  // Se localiza la variable en el Bank 1
  dato_bank2 : word absolute $120;  // Se localiza la variable en el Bank 2
  dato_bank3 : word absolute $1A0;  // Se localiza la variable en el Bank 3
begin
  {$MSGBOX 'BANCO DE MEMORIA ACTUAL = ' + CURRBANK}
  dato_bank0 := word({$RESULTADO}); 
  {$MSGBOX 'BANCO DE MEMORIA ACTUAL = ' + CURRBANK} 
  dato_bank1 := word({$RESULTADO}); 
  {$MSGBOX 'BANCO DE MEMORIA ACTUAL = ' + CURRBANK} 
  dato_bank2 := word({$RESULTADO}); 
  {$MSGBOX 'BANCO DE MEMORIA ACTUAL = ' + CURRBANK} 
  dato_bank3 := word({$RESULTADO}); 
  {$MSGBOX 'BANCO DE MEMORIA ACTUAL = ' + CURRBANK} 
end.
```
Primero darte la enhorabuena por el control actual sobre los bancos de memoria RAM donde se sitúan las variables.

Y el detalle de los decimales basura se da cuando la velocidad de reloj es, por ejemplo, 12 MHz. En lugar de obtener en la ventana de mensajes el dato 0,333333333333333333, a partir del decimal 9 no es correcto, y si pones 10 MHz se debería obtener 0,4 los "decimales basura" estropean el resultado a partir nuevamente el noveno decimal.

Ya se que no es algo significativo, pero estaba probando otros que funcionan perfectamente, y este detalle estético creo que también hay que cuidarlo. :)

t-edson, 2017-09-09:
Sí, lo había notado. Es un tema de redondeo inherente a los cálculos en coma flotante, del intérprete de directivas (Ese nombre es muy largo. Mejor lo llamo IDD o DII). Sé que se puede solucionar, redondeando la precisión del resultado, pero no le habíada dado mucha importancia, porque como mencionas, no es crítico, y además, está pendiente de implementarse, algunas funciones de manejo de números y de cadenas, que deberían solucionar este tema.

t-edson, 2017-09-18:
En la versión 0.7.7, se ha mejorado la precisión de los cálculos con números decimales. Así que este problema debería estar resuelto. Además se han creado las funciones round() y trunc(), que solucionarían algunos casos de redondeo.

AguHDz, 2017-09-20:
Ok, confirmado. En versión 0.7.7 el problema está resuelto.

closed 2017-09-20 by AguHDz