t-edson/PicPas #19: I want to say you some bug report, but i didnt find a way for that.

tikitatiktik, 2017-11-21:
you have an error "for" loop and "while  TXSTA_TXEN = 0 do end; " . Those are very important. I want to inform you, if you are interest. Good luck. You will see when you have to try this command row for test.

t-edson, 2017-11-22:
Hi. Could you please give more detail on the error. I can compile "while TXSTA_TXEN = 0 do end; " without problems.

tikitatiktik, 2017-11-26:
Hi. You will see an error when you focus to highlighted lines on the picture.

picpas:

while  TXSTA_TRMT = 0 do end;

list file:

0x0DD btfsc RCSTA_OERR

"TXSTA" is converting to "RCSTA" .  Thanks for your interest. Have a nice day.

________________________________
Gönderen: Tito Hinostroza <notifications@github.com>
Gönderildi: 22 Kasım 2017 Çarşamba 14:35:47
Kime: t-edson/PicPas
Bilgi: tikitatiktik; Author
Konu: Re: [t-edson/PicPas] I want to say you some bug report, but i didnt find a way for that. (#19)

Hi. Could you please give more detail on the error. I can compile "while TXSTA_TXEN = 0 do end; " without problems.

tikitatiktik, 2017-11-26:
[image]

t-edson, 2017-12-01:
I see. The code is compiled right. The assembler code is wrong. TXSTA and RCSTA share the same address but different RAM bank. The disassembler, have no way to know if the address 0x18 of the opcode btfsc, refers to teh bank0 or the bank1.
The compiler work well. It's just the assembler view is generated from the binary directly, where the bank address is lost.

t-edson, 2017-12-01:
PicPas doesn't generate assembler code. It compiles directly to binary. That's the problem. Once binary created, there is not information about banks in opcodes, in the binary. It's an issue of the disassembler. 
I have considered to include some aditional information in the binary, to solve this problem, but it will take some time.

closed 2017-12-27 by t-edson