t-edson/PicPas #22: error on function implementation

alexpseletr, 2018-01-05:
function adc_read(adc_chan:byte):word;

error : ":"  or ", " expected.

t-edson, 2018-01-05:
PicPas use the syntax of Moula-2 no standrad Pascal . To declare functions, you use the same word PROCEDURE. So you declare a function:

```
procedure adc_read(adc_chan:byte):word;
begin
end;

```

alexpseletr, 2018-01-06:
your solution worked only as a suggestion the project (http://www.pas2pic.hu/) that works only at command prompt accepts the expression function

t-edson, 2018-01-06:
It's not a problem to implement the FUNCTION keyword. It's just PicPas bet on a modern Pascal syntax. The same is true for structures like IF, WHILE and FOR. There is "Pascal mode" in PicPas that will accept the classic Pascal syntax.

closed 2018-01-06 by alexpseletr