t-edson/PicPas #23: suggestion of folder structure

alexpseletr, 2018-01-06:
as the project grows and you have more devices and examples I believe that there are many files in the unit folder that are difficult to find ...

suggestion:
create folder devices for device files and keep the functions in the folder unit or another with another name ...

t-edson, 2018-01-06:
Yes. That's a good suggestion. I have considered to include the folder "/devices" but by the moment, there is just a few units. Maybe you consider to collaborate with new units.

alexpseletr, 2018-01-06:
I'm working on some modifications and launching in my git, logically appears in real time for you, you just need to accept the commits

alexpseletr, 2018-01-06:
could you tell me where to define new models in the lazarus source code? I'm looking for 2 days and I'm not finding, I just found the ones in the folder units

t-edson, 2018-01-06:
Excelent. I will check the commits before to include. 
The files in the folder "/units" are just units to help on having the names for the SFR of the RAM and for some useful definitions like the Configuration bits.
But they are supposed to work after using {$PROCESSOR <device>} so the compiler needs to know internally the device.

closed 2018-01-06 by alexpseletr

t-edson, 2018-01-07:
I see you have included the unit for PIC12F679.  But I cannot find a datasheet for this Model. Maybe it's a error in the name,