t-edson/PicPas #24: unit PIC16f877A error line 244 EEDATA invalid memory address

alexpseletr, 2018-01-06:
in the datasheet the address is correct

t-edson, 2018-01-06:
Do you refer the unit PIC16F877A.pas has a wrong address?
What is the wrong address?  What is the right address?

alexpseletr, 2018-01-06:
he compiler points out an error in
  EEDATA: byte absolute $ 010C;

silvioprog, 2018-01-06:
The declared address is OK, since XC8 headers also declares it as 0x10c:

```c
...
#define EEDATA EEDATA
extern volatile unsigned char           EEDATA              @ 0x10C;
...
```

So, you need to send some trace log to a deep checking.

t-edson, 2018-01-06:
The address $010C. Maybe you refers to the error when trying to compile the unit PIC16F877A: 
"Invalid address for this device".
It is "normal", because the unit PIC16F877A doesn't indicate to the compiler how many RAM exist in the current device.
Anyway in a normal project you are not going to compile PIC16F877A alone. You just include it as part of a program:

```
program Name;
{$PROCESSOR PIC16F877A}
uses PIC16F877A;
...
```

t-edson, 2018-01-06:
Other option I'm considering is to modify the unit PIC16F877A.pas and include the directive:

`{$PROCESSOR PIC16F877A}`

at the beginning. So you can compile the unit without errors. And moreover, you won't need to use always this directive, in your program.
The rule in PicPas is: 

"The compiler doesn't know wich hardware is using until found the directive {$PROCESSOR]"

Before of that PicPas assume a default device, very similar to the PIC16F84.

silvioprog, 2018-01-06:
+1 for modifying `PIC16F877A.pas` with `{$PROCESSOR PIC16F877A}`.

alexpseletr, 2018-01-06:
Sorry, it was my mistake here. I did not save the pic model.

closed 2018-01-06 by alexpseletr