t-edson/PicPas #25: [QUESTION] 877A support

silvioprog, 2018-01-06:
Hi.

Do you have plan to maintain 877A support? I've looked at Microchip's news and got the following (bad) news:

> PIC16F877A 
> **Not Recommended for new designs**
> *Please consider this device PIC16F18877*
> [More...](http://www.microchip.com/wwwproducts/en/PIC16F877A)

However, the 18877 models are very recent, expensive, and hard to be found in common sites (aliexpress, ebay etc.).

Thank you!

t-edson, 2018-01-06:
The old devices (like this or the PIC16F84) are maintained for historical reasons, for testing and because there is still people using this devices. 
Personally, I prefer to use the 16F628 or the PIC16F887 in my designs. They are easy to obtain here.
PicPas doesn't have libraries for all PIC supported. The few that exists has been created for people who wanted to collaborate. 
I'm more dedicated to extend the compiler features than in create libraries. If you need to use another device maybe you can create a new library for the device and I will be glad to include it in PicPas.

t-edson, 2018-01-06:
By the moment, PicPas only support the classic 14 bits core, and this include a lot of devices. 
Not of all 14 bits devices are directly supported by PicPas. It has some "internal" definitions for just a few, but the compiler include the mechanisms to define "by program" any other 14-bit core device.

That is, if you want to use the PIC16F887 you just have to use the directive:

{$PROCESSOR PIC16F887}

And the compiler will have the hardware description of the device.

But if you want yo use some like the PIC16F917, the compiler won't have the description internally, and you would need to define the hardware using directives. This is explained in the User Manual.

t-edson, 2018-01-06:
I have plans, this year, to include support for the series 14-bits Enhaced Midrange devices and for the 18F series. It depends of my free time and of the collaboration I get from other people.

closed 2018-01-11 by t-edson