t-edson/PicPas #27: error maximum number of RAM

alexpseletr, 2018-01-07:
I do not know if it is some error of mine, but in pic 12f675 it shows 128B of maximum ram, that value is that of ram eprom is 64 and in 16f877A it shows 416 in datasheet 368

may not be anything, but I'm reporting to improve the app

I found some settings in pic16Devices.pas but I did not succeed in fixing

t-edson, 2018-01-08:
I have seen it before. I remember it was an issue about the procedure that make the calculus. It was considering the common RAM too. I will check.

t-edson, 2018-01-11:
Algorithm to calculate the available RAM has been fixed. Please check.
PicPas is using now the library PicUtils in version 1.2 (changes made here).

closed 2018-01-21 by t-edson