t-edson/PicPas #28: improvement in libs

alexpseletr, 2018-01-07:
I intend to make an improvement in the libraries by putting error messages if something is set wrong

example if pin3 = output then {$ MSGBOX 'error config pin3'}

so a pin that can only be used with input if it is used as output (this in the user's design unit) error appears when compiling ...

I do not know, it's possible or it's viable.

t-edson, 2018-01-08:
Do you refer to set a pin as output (TRISA.5 = $0, or SetAsOutput() ) but it can only be setted as Input (like MCLR)?
Ot you refer to use a pin (maybe with ORTB.1 = 0) that has been previously defined (maybe with TRISB ;= $FF or SetAsInput() ).

alexpseletr, 2018-01-08:
It is not case to use the pinA3 of the 12f675A as output ua as it can be used as input or other problems that are still in progress ... hence I would already put the locks on the libs that would help the novatoes or at a time of lack of attention

t-edson, 2018-04-08:
Reading the value of bit of RAM is not possible in compilation time. There are many ways memory RAM can be changed and it's not factible the compiler analyze all the posibilities.
Maybe it can be prevented to set a worng value when using the SetAsInput/SetAsOutput instructions if they were macros. But by now PicPas macro support cannot do that.

closed 2018-04-13 by t-edson