t-edson/PicPas #31: error apparent function SetAsInput and SetAsoutput no TRISA action

alexpseletr, 2018-01-08:
I'm using the code below to test it.
1) TRISA is not changed in the assembler
2) trisB manually in asm it appears as PORTB, in the simulator the TRISB is triggered (in trisB it should be only the name of the variable)

program NewProgram;
{$PROCESSOR PIC16F84A}
{$FREQUENCY 4MHZ}
uses PIC16F84A;
  
//Declarations here
  
begin
  
  SetAsInput(PORTA_RA0);
TRISB:=$FF;

  
end.

===========asm part============

TRISA EQU 0x000; <Unused>
TRISB EQU 0x086
direccion EQU 0x000; <Unused>
valor EQU 0x000; <Unused>
addr EQU 0x000; <Unused>
;------ Work and Aux. Registers ------
;===Blocks of Code===
    0x000 goto 0x001 
__main_program__:
    0x001 bsf STATUS, 5 ;Bank set.
    0x002 bsf PORTA_RA0 
    0x003 movlw 0xFF 
    0x004 movwf PORTB 
__end_program__:
    0x005 sleep  
;--------------------
    END 

t-edson, 2018-01-08:
I see the ASM code is right. It's an issue of the disassembler.  Take a look to https://github.com/t-edson/PicPas/issues/19
If you simulate the code, you will see TRISA and TRISB are changed. It's just the disassembler doesn't know that the current bank is 1, when found the Opcode BSF, because the PIC ASM Opcodes don't have bank information.

alexpseletr, 2018-01-08:
TRISA also does not appear in the simulator, in Proteus I also had strange operations in some test codes ...

maybe we should change the way that SetAsInput and SetAsout get the parameters ... or something like that

t-edson, 2018-01-08:
Why TRISA doesn't appears (and why appears some  other unused registers), is another issue, maybe a strange feature. Anyway you can add it, manually (if cannot, it's a bug). 
The simulator is still experimental so there is much work to do in it. PicPas simulator only simulate the CPU core.
Proteus has some bugs too. Probably there are differences in execution.

t-edson, 2018-04-08:
This problem should be solved in the version 0.8.4. 
[image]

closed 2018-04-08 by t-edson