t-edson/PicPas #32: suggestion of implementing new macros

alexpseletr, 2018-01-08:
define
 on = 1
off = 0
high=1
low=0
true=1
false=0

bitisSet
bitisClear

some below that I found on the net

;********CONFIGURACION DE PINES DEL MICRO****************

;-------------PIN MODE OUTPUT---------------------
;Configura cualquiera de los pines como SALIDA
;-------------------------------------------------	
_pin_m_out	    macro   TRISX,PORTX,LATX,ANSELX,RXX
		    banksel PORTX
		    bcf	    PORTX,RXX
		    banksel ANSELX
		    bcf	    ANSELX,RXX
		    banksel TRISX
		    bcf	    TRISX,RXX
		    endm

;-------------PIN MODE INPUT---------------------
;Configura cualquiera de los pines como ENTRADA
;------------------------------------------------		    
_pin_m_in	    macro   TRISX,PORTX,LATX,ANSELX,RXX
		    banksel PORTX
		    bcf	    PORTX,RXX
		    banksel ANSELX
		    bcf	    ANSELX,RXX
		    banksel TRISX
		    bsf	    TRISX,RXX
		    endm
		    
;-------------PIN HIGH---------------------------
;Coloca un estado ALTO en el PIN seleccionado
;------------------------------------------------		    
_pin_hi		    macro   TRISX,PORTX,LATX,ANSELX,RXX
		    banksel LATX
		    bsf	    LATX,RXX
		    endm
		    
;-------------PIN LOW---------------------------
;Coloca un estado BAJO en el PIN seleccionado
;-----------------------------------------------		    
_pin_lo		    macro   TRISX,PORTX,LATX,ANSELX,RXX
		    banksel LATX
		    bcf	    LATX,RXX
		    endm
		    
;-------------PORT MODE OUTPUT---------------------
;Configura un PUERTO completo como SALIDA
;-------------------------------------------------		    
_port_m_out	    macro   PORTX
		    banksel PORTX
		    clrf    PORTX
		    banksel ANSELX
		    clrf    ANSELX
		    banksel TRISX
		    clrf    TRISX
		    endm

;-------------PORT MODE INPUT---------------------
;Configura un PUERTO completo como ENTRADA
;-------------------------------------------------		    
_port_m_in	    macro   PORTX
		    banksel PORTX
		    clrf    PORTX
		    banksel ANSELX
		    clrf    ANSELX
		    banksel TRISX
		    movlw   B'11111111'
		    movwf   TRISX
		    endm
		    
;******************************************************	

;********MANEJO DE REGISTROS DEL MICRO*****************
		    
;-------------FILE REGISTER BIT HIGH-----------------------
;Coloca un estado ALTO en el BIT del REGISTRO seleccionado
;----------------------------------------------------------		    
_fb_hi		    macro   file,bit
		    banksel file
		    bsf	    file,bit
		    endm
		    
;-------------FILE REGISTER BIT LOW------------------------
;Coloca un estado BAJO en el BIT del REGISTRO seleccionado
;----------------------------------------------------------		    
_fb_lo		    macro   file,bit
		    banksel file
		    bcf	    file,bit
		    endm

;-------------FILE REGISTER LITERAL------------------------
;Carga un valor LITERAL en el REGISTRO seleccionado
;----------------------------------------------------------		    
_fb_li		    macro   file,value
		    banksel file
		    movlw   value
		    movwf   file
		    endm

;**********************************************************		    

;-------------IF PIN HIGH GOTO LABEL------------------------
;SI el PIN se encuentra en ALTO GOTO label
;-----------------------------------------------------------		    
_if_pin_hi_go	    macro   TRISX,PORTX,LATX,ANSELX,RXX,label
		    pagesel label
		    banksel PORTX
		    btfsc   PORTX,RXX
		    goto    label
		    endm
		    
;-------------IF PIN LOW GOTO LABEL-------------------------
;Si el PIN se encuentra en BAJO GOTO label
;-----------------------------------------------------------		    
_if_pin_lo_go	    macro   TRISX,PORTX,LATX,ANSELX,RXX,label
		    pagesel label
		    banksel PORTX
		    btfss   PORTX,RXX
		    goto    label
		    endm
		    
;-------------IF FILE REGISTER BIT HIGH GOTO LABEL----------
;Si el BIT del FILE REGISTER se encuentra en ALTO GOTO label
;-----------------------------------------------------------
_if_fb_hi_go	    macro   file,bit,label
		    pagesel label
		    banksel file
		    btfsc   file,bit
		    goto    label
		    endm
		
;-------------IF FILE REGISTER BIT LOW GOTO LABEL----------
;Si el BIT del FILE REGISTER se encuentra en BAJO GOTO label
;-----------------------------------------------------------		    
_if_fb_lo_go	    macro   file,bit,label
		    pagesel label
		    banksel file
		    btfss   file,bit
		    goto    label
		    endm

t-edson, 2018-01-08:
Do you refer to:
1. Implement MACROS inside the ASM blocks, or 
2. Implement new Pascal instructions to acomplish this functions? 

IN CASE 1: Do you refer to have this macros predefined or implement the ability to define macros inside ASM blocks.
IN CASE 2: Do you refer to extend the DIRECTIVES functionality to create macros with parameters or do you refer to have instructions equivalents to these macros?

alexpseletr, 2018-01-08:
yes, pre-defined macros as you already have delay_ms (); or still SetAsInput () ... I have not found it yet in the code, but if I indicate in the sources I can try to help in the implementation

t-edson, 2018-01-09:
delay_ms() and SetAsInput() are called "System function" in PicPas. They are not macros. Macros are defined using directives ($DEFINE or $SET).
System functions are created in several points, mainly in unit "GenCod.pas":
1.-> Unit "GenCod.pas", method TGenCod.StartSyntax(): Define the name of the system function, as identifier of attribute "tnSysFunct".
2.-> Unit "GenCod.pas", method TGenCod.CreateSystemElements(): Define the system function and refer to a implementation method. 
3.-> Unit "GenCod.pas": Define a implementation method of the system function. Something like TGenCod.fun_SetAsInput(). Here we decide if the implementation is INLINE or a call to a subroutine.
4.-> To activate the highlighter and completion code of the new system function it's needed to modify the file PicPas_PIC16.xml.
5.-> To assure the new system fucntion works properly. a test code must bee added to the test programs.
6.-> Finally it's needed to update the User manual and the GitHub Readme.txt.

closed 2018-04-08 by t-edson