t-edson/PicPas #34: Error example LCD_Test.pas

moonrockets, 2018-01-22:
Last few lines of example LCD_Test.pas

    until Counter = 0;
 
    LcdCommand(LCD_CLEAR_DISPLAY);
  end;
end.
///*************************************************************//174

not working this from source-github.... 

    LcdCommand(LCD_CLEAR_DISPLAY);
  end;
**end;**  <======= ***** this inserted, compilation succesfull ....
end.
///*************************************************************//174

only the **END;** inserted and compilation successfull -- any idea why?

moonrockets, 2018-01-22:
failed:
////////////////////////////////////////////
// New program created in 22.01.2018}
////////////////////////////////////////////
program NewProgram;
{$PROCESSOR PIC12F629}
{$FREQUENCY 4MHZ}
  
//Declarations here
begin
  //Code here
  while true do 
   begin 
   end;
end.

bug in the while-construct?
insert **end;** above the END. ---> compilation okay

t-edson, 2018-01-23:
The structure WHILE in PicPas have the syntax: 

```
  while true do
    //while body
  end;
``` 
This is different from the classic Pascal WHILE.

moonrockets, 2018-01-23:
sir - may i give impression to be stolid ...
i got confused about the while-construct because load your example LCD_test.pas from github-folder
PicPas/samples/LCD_test/LCD_test.Pas (files uploaded may-26-2017)

okay, i got your answer, ... just close my issue  ;)

t-edson, 2018-01-23:
I see the error. The sample /PicPas/samples/LCD_test was written for an old version of PicPas, whe the default syntax was the classic Pascal. Now the default syntax is the Modula-2 style. 
Anyway, it's fixed now.

closed 2018-02-07 by t-edson