t-edson/PicPas #37: Need help ?

cuistax, 2018-04-07:
Hello all,

I just discovered your project and find it very interesting. 

I would be happy to contribute, I think I could be helpful for:
- Mac target binaries
- French translation (and English)
- Support for new PICs

Besides, I have one question, a friend of mine is using Linux. What's the most recent version he can use ? Any plan to release up to date binaries for that platform ?

Thanks!

t-edson, 2018-04-07:
Hi.

Thank for test. All collaborations are wellcome. French translation would be excellent for PicPas. Just tell me to add new entries for a new language. And tell me if you have doubts about adding new PICs. It's needed too, to complete the current PIC definitions.

I mostly works only on Windows so I don't compile Linux version for all versions of PicPas.  Just I compile a binary for Ubuntu in some versions. Tha last was for the 0.8.2 version. Anyway I can compile a Ubuntu binary of the currrent version if it's needed.

I don't have a Mac to compile so I cannot test on that. But if you can compile it would be great.

cuistax, 2018-04-08:
Ok great.

I will look into that. I will get back to you once I get some result.

t-edson, 2018-04-08:
A binary Ubuntu version is compiled for PicPas 0.8.4: 
[image]

cuistax, 2018-04-08:
Excellent, thanks. I will install that version on my friend's Linux.

I am currently trying to compile the project on my Mac. 
So far the error messages are winning, but the fight is barely started ;)

cuistax, 2018-04-09:
Hey,

I could get it to work on Mac, where shall I put the binary ?

[image]

Please note that I had to update Misutils.pas: it seems that with Lazarus 1.8.2 (latest version), unit FileUtil is not supported anymore, it must be replaced by LazFileUtils.

Thanks,

t-edson, 2018-04-10:
Excelent. Please put the binary in your GitHub or put it directly here in the PicPas repository and I will commit the request.

cuistax, 2018-04-10:
There you go. As I stated in the install note, it must yet be considered experimental. 

Many details, such as the app icon, have yet to be sorted.
That said, the app seems to work quite well on my Mac (isn't Lazarus portability great ??)

So, the next release can only be better ;)

t-edson, 2018-04-10:
Thanks. I will compile the next version using Lazarus 1.8. If you found some details in the code that need to be changed in order to easy the compiling to Mac, just tell me to change. 
Some features like the file explorer, works only on Windows. They need to be extended to Linux and mac too. All other features should work OK in all platforms.

cuistax, 2018-04-19:
Regarding your last comment, at the moment two things can be updated:
- the use of LazFileUtils instead of FileUtil in Misutils.pas (as mentionned in post nr. 6)
- the project name should be renamed from "project1" to "PicPas" because, as you can see in the Mac screenshot (also in post nr. 6), the name "project1" gets displayed by default in the menu bar.
I can build another Mac binary whenever needed.

t-edson, 2018-04-19:
OK. I will do the changes.

t-edson, 2018-04-19:
Ready. New project is now "PicPas", compiled with Lazarus 1.8 and the unit FileUtil is changed. 

This version 0.8.5 use now the library https://github.com/t-edson/ogEditGraf in the version 2.2 (the default), so it needs to be updated to compile the project PicPas.

t-edson, 2018-05-01:
Many internals changes have been done in the version 0.8.5. I'm closing this version to open a new version.  I'm geenrating the exe for Windows-64 and Ubuntu. Please,generate the binary for Mac in this version. Don't forget to update the library PicUtils.
I hope the 0.8.6 will include finally a compiler for the Baseline family. Let's see what can be done.

cuistax, 2018-05-04:
Yeah of course, but I can't seem to compile the project right now.

According to the error log below, there is a file missing:

```
(3104) Compiling FrameArcExplor.pas
/tmp/PicPas/Source/FrameArcExplor.pas(167,4) Fatal: (2013) Cannot open include file "../language/tra_FrameArcExplor.pas"
```

t-edson, 2018-05-04:
It's my fault. I have created a new translation file. Maybe you can add the french translation to this new file. It's uploaded now.

t-edson, 2018-05-05:
I have opened the branch 0.8.6 and have the new changes there. Please make the changes on that version.

cuistax, 2018-05-05:
The Mac binary for branch 0.8.5 is ready.

t-edson, 2018-05-09:
Thanks. I'm glad to say the version 0.8.6 can additionally use the PIC10 compiler so it can compile now for the 12 bits core devices. There are still, some forms to modify to adapt to these new compiler and the simulator is not working by now, but PicPas can now compile a "Hello World" for the PIC10F200.

cuistax, 2018-05-14:
I would like to compile version 0.8.6 but I get the following errors:

````
FormDebugger.pas(114,7) Error: (5038) identifier idents no member "ExecNCycles"
FormDebugger.pas(115,6) Warning: (5036) Local variable "stopped" does not seem to be initialized
FormDebugger.pas(161,16) Error: (5038) identifier idents no member "DisassemblerAt"
FormDebugger.pas(162,15) Error: (5038) identifier idents no member "ReadPC"
FormDebugger.pas(224,15) Error: (5038) identifier idents no member "ReadPC"
FormDebugger.pas(262,14) Error: (5038) identifier idents no member "DisassemblerAt"
FormDebugger.pas(281,7) Error: (5038) identifier idents no member "Reset"
FormDebugger.pas(295,7) Error: (5038) identifier idents no member "Exec"
FormDebugger.pas(297,7) Error: (5038) identifier idents no member "ExecNCycles"
FormDebugger.pas(298,6) Warning: (5036) Local variable "stopped" does not seem to be initialized
FormDebugger.pas(323,7) Error: (5038) identifier idents no member "WritePC"
FormDebugger.pas(338,7) Error: (5038) identifier idents no member "ExecTo"
FormDebugger.pas(365,7) Error: (5038) identifier idents no member "ExecStep"
FormDebugger.pas(371,7) Error: (5038) identifier idents no member "Exec"
FormDebugger.pas(474) Fatal: (10026) There were 12 errors compiling module, stopping
````

t-edson, 2018-05-14:
It looks like you have to update PicUtils 1.4. Several changes have been done there.
I just updated PicPas 0.8.6. Now the compiler include a Combo in the IDE to manually switch the compiler to use. The units for devices are separated in /devices10, /devices16, /devices17 (empty) and /devices18 (empty). And the Debugger can now simulate too the 12 bits PICs.
Tell me if you have problems compiling.

cuistax, 2018-05-28:
Did you find a solution to that ?

t-edson, 2018-05-28:
Hi. I'm working on that. There are thousands lines of code to check, identify and change. It's not only for the places where FSR is used; but when the instruction ADDLW or SUBLW are used too. Now the work is in the version 0.8.7 and several changes have made in the internal architecture of the classes to reuse most of the code of the compilers.

t-edson, 2018-05-31:
Maybe you can collaborate creating the Test code for the Baseline compiler and test them. The files are in /testcode10.
I have an additional problem. I use Proteus to run simulation for the Test code and I have an old version (and cannot update)  and cannot simulate a multi-RAM bank device like the PIC16f57. This limit my possibilities to test the new PIC10 compiler.

cuistax, 2018-05-31:
Ok I will look into that.

On my side I have several (real) baseline chips that I can use for testing : PIC10F202, PC12F508 and PIC16F506.  

As for the simulation part, you may be interested to know that with Microchip's MPLAB X IDE, you can import external .hex files into a new project and run it on their simulator.
And this is possible with the free version, you don't have to pay anything.

This video from Microchip shows how to import external .hex files : [https://www.youtube.com/watch?v=pEMORwwuyos](https://www.youtube.com/watch?v=pEMORwwuyos).
In this video they only program a chip, but you can simulate it also.

I have already done it with .hex files generated by the SDCC compiler, it works.
Let me know if you need further information on that.

t-edson, 2018-06-01:
I can test some routines in the same simulator of PicPas and it's the more comfortable way to test. But the PicPas simulator needs to be "validated" before using some other Simulator. Moreover, there is not an interfaz so powerful like Proteus have. Maybe it's better to improve the PicPas simulator and make all the tests run in just one click.

cuistax, 2018-06-01:
> Check the Proteus circuit in /testcode10.

Where is it ?

> The idea would be to automate all the tests in a simple Click.

> The problem is the test codes of PicPas is prepared to run in Proteus

> I can test some routines in the same simulator of PicPas

So, if I follow you:
- your tests are currently only usable with Proteus
- you would like to validate both the PicPas simulator and the generated code, but currently only the generated code can be tested
- you would like to use the PicPas simulator to validate the compiled code, and get rid of Proteus
- you would like to get better automation for tests

Is that right ?

So, just to clarify things for me, could you answer the following questions:

1) how exactly would you like to validate the simulator ? what would the process be like ? what is currently missing to achieve that goal (in terms of dev, not in terms of test cases) and how much effort would it require ? 

2) how exactly would you like to validate the compiled code ? what would the process be like ? what is currently missing to achieve that goal and how much effort would it require ?

t-edson, 2018-06-01:
> Where is it ?

I haven't uploaded the folder. It's updated now. 
Here I have prepared a set of test in the same way it exists for the PIC16 compiler in the folder "/TestCode" ( I will change the name to "/testcode16" later) These test are some extense but they are far to be complete.

>    your tests are currently only usable with Proteus

The short answer is YES. They were created for Proteus to test the first versions of PicPas. Maybe these test can be used too in MPLAB but it will take much more time to verify (in the way they are written now). Probably hours. In Proteus they can be tested in minutes.
There is a group of test that can (must) be verified directly in PicPas, like https://github.com/t-edson/PicPas/blob/0.8.7/TestCode/MustPass/Test5_Banks.pas, that use DIRECTIVES to verify the internal variables of the compiler

>    you would like to validate both the PicPas simulator and the generated code, but currently only the generated code can be tested

Debugger/simulator was one of the early objectives of PicPas. It's part of the IDE and like any other tool it must be correctly tested. I have tested by comparing the results when executing code and I must say the PicPas Debugger/simulator is very accurate (I haven't report for bugs). Even it's, in some cases, more exactly than Proteus, but that's another issue. 
The PicPas simulator is very exactly emulating all the instructions of the PIC CPU Core, but it lacks of simulation for pheriphericals. Timers, Watchdog, UART, ... are not currently simulated. There is some work to do to implement this, and this is compiler developing, this is something different.
 
However I have not a formal way to test the Simulator. I think a formal test must not be so complicated, because PIC Cores are RISC, and it would be focused in check the result for all the instructions (in results and in cycles used)

>    you would like to use the PicPas simulator to validate the compiled code, and get rid of Proteus

Yes.

>    you would like to get better automation for tests

Yes.

cuistax, 2018-06-01:
Well, to start somewhere, I believe the first necessity would be to have a test framework for the simulator.

At the very least, all you need is a function which takes in input:
- an initial memory content (code+data)
- an expected memory content (data only) after the input program has run

The function takes the input code+data, runs it into the simulator (the code must terminate itself), and once simulation is done, compares the obtained memory content with the expected one, and returns a validation status.

You can always refine it later to take the number of cycles into account and that sort of things, but that would get you started.

How much effort would that take ?

t-edson, 2018-06-02:
Basically the test for Simulator would be one more program to simulate. The idea would be to use all the instructions and check the result, the state of the flags, and the cycles used. It' would be interesting doing a consistent program using all the instruction of one family. The test should include invalid Opcodes too, to verify the response.
I think this will require to add some features to the simulator like components (similar to a led)s that can be conected to a pin of the PIC and generate some control action (like  Stop Simulation, Reset device, Show a Message box, write a text in the message window, generate a Sound, ... ).
[image]
This is a lot of work (some kind of visual programming) but I think this is the best way to proceed.
It's projected to include some LCD and Serial Terminal to the simulator but this devices are jus part of the hardware simualtion and what we need is a better control of the simulation.
The detection of errors will: 
* Stop the simulation
* Locate the point of the simulation where it fails (and better the point of the source code).

Doing all this hard work it's only the first phase of automation testing (assuming we have migrated all the test code) and will let us prescind of Proteus. 
The next phase would be the real automation that is the possibility to run several programs with just one click or command. This would involucrate some kind of script or code that simulate sequentially many files and stop when some fails, showing the file and the point where the test failed.

t-edson, 2018-06-10:
Hi. I've added several functions to the simulator of PicPas. Now it's possible to change the state of pins of the microcontroller and connect the output to simple components: 
[image]
Several changes has been done in this version. I'm going to close the 0.8.7. Please generate the executable for Mac. You need to update https://github.com/t-edson/PicUtils and https://github.com/t-edson/ogEditGraf

cuistax, 2018-06-11:
That looks really pretty !
I will check it out.

And yes, I will provide an updated Mac binary soon...

closed 2019-07-01 by t-edson