t-edson/PicPas #41: Compiler issue with numeric constants

cuistax, 2018-04-16:
Hello,

I am encountering compiler issues with numeric constants, for instance:

```pascal
const BASE_DELAY      = 100; // => compiles ok [byte]
```

```pascal
const BASE_DELAY      = 300; // => compiles ok [word]
```

```pascal
const BASE_DELAY      = 100;
const EXTENDED_DELAY  = BASE_DELAY * 3; // => compiles ok
```
```pascal
const BASE_DELAY      = 300;
const EXTENDED_DELAY  = BASE_DELAY * 3; 

// => COMPILER ERROR: Undefined operator: * for type: word
```
```pascal
const BASE_DELAY      = 300;
const EXTENDED_DELAY  = 3 * BASE_DELAY;

// => COMPILER ERROR: Illegal Operation: (byte) * (word)
```
So it seems there is an issue with the handling of mixed byte and words sizes.

t-edson, 2018-04-16:
It's because the operations (word)*(byte) and (word)*(byte) are not implemented in the compiler. Not even for constants. It coud be implemented only for constant (no need to create ASM code) but the best solution should be implement this operations in the compilers. PicPas lack of some operatioons for WORDS and DWORDS.

t-edson, 2018-04-20:
It's solved in the version 0.8.5. Implemented multiplication for constant. It's still needed to implement multiplicaction for Word a Dword data types in the code generator.

cuistax, 2018-04-23:
I have re-tested the cases that raised an error in my first post. 
I confirm that they successfully compile now.

closed 2018-04-23 by t-edson