t-edson/PicPas #43: French translation

cuistax, 2018-04-19:
Hello,

I should now be able to work on the french translation.
Can you describe me the procedure to follow ?

(I saw the "language" folder, but I'm not sure how to proceed) 

Thanks,

t-edson, 2018-04-19:
OK.  I need to create a new entry for a new language. Then you can add the translation. I will tell you when ready.

t-edson, 2018-04-19:
Ready. A new entry is created to add the french translation. Just add the translation in the las parameter of the function Trans() in all the files *.pas of the folder /language.

The english language is configurated as the default language when a french transaltion is not found.

cuistax, 2018-04-20:
Hey,

French translation has been added here : #45 

In the process I did identify a couple of small issues:

1) I could not translate some parts of PicPas, like the "RAM Explorer" menu item:
[image]
 
2) the left pane of the Settings window crops items, it should be enlarged:
[image]

3) still in the Setting window but not related to translation, I did notice that in the left pane, for some items a **simple** click is enough, but for certain items a **double** click is necessary to show the corresponding right pane. Do you experience this  also ?

4) also un-related to the translation but maybe more of a functional issue, in the Settings dialog,  then External Tools item, if I click the the little icon to select a binary app and select any app on my disk, the "app folder path" text does not get updated. Is it only happening on Mac or do you also experience it ?

5) the simulator/debugger dialog does not seem to have been translated at all. Is it correct ?

t-edson, 2018-04-22:
> I could not translate some parts of PicPas, like the "RAM Explorer" menu item:

Yes. There are still some parts of the code to translate. I have including this text and some others in the file tra_FormPrincipal.pas but it's needed to including other strings for the other files.

> the tree pane of the Settings window does crop tree items, I think it should be enlarged:

Fixed.

> still in the Setting window but not related to translation, I did notice that in the tree pane, for some items a simple click is enough, but for certain items a double click is necessary to show the corresponding right pane. Do you also experience this ?

No. No in Windows nor in Linux. Maybe an issue with Mac. Anyway I cannot test. Maybe you can find in the code for Mac.

>also un-related to the translation but maybe more of a functional issue, in the Settings dialog, then External Tools item, if I click the the little icon to select a binary app and select any app on my disk, the "app folder path" textbox does not get updated. Is it only happening on Mac or do you also experience it ?
No. In Windows it works correctly. Maybe it's a bug in Mac.

>the simulator/debugger dialog does not seem to have been translated at all. Is it correct ?

You're right. It has not a translation. It's necessary to create the file tra_FormDebugger.pas and include in it the string translations.

>It seems the shortcuts are not correctly handled. In the example below, the shortcut for opening a file -in french- should be "^O" (according to the source file I changed), but it has been assigned "^A" (according to the screenshot) . Is seems that somehow it keeps using the spanish shortcut.

The shorcut assigned in the string only works when using the menú with keyboard. The keystroke <^A> is assigned statically in the project. I have not yet a way to set this in the language file.