t-edson/PicPas #55: Missing some SET_STATE_RAM

Lulu04, 2018-06-13:
Hi, 
Cuistax, you did a lot of work on PIC declaration files !
I have checked the PIC16F1825.pas in 'devices17' and it seems to miss  the SET_STATE_RAM declaration for the core registers in each bank. Only the INDF register is declared. The SET_MAPPED_RAM section semms to be well defined.

cuistax, 2018-06-13:
Hello Lulu04 !

What file are you looking at ?

The latest file is at [https://github.com/t-edson/PicPas/blob/0.8.7/devices17/PIC16F1825.pas](https://github.com/t-edson/PicPas/blob/0.8.7/devices17/PIC16F1825.pas)

The SET_STATE_RAM directives start at line 678.

Let me know if anything is missing.

t-edson, 2018-06-13:
I can confirm it. If we look at the RAM explorer after compiling PICF1825, we have:
[image]

It's some difficult to see 32 banks in one window, but it can be apreciate the upper bytes of banks 1 to 31 are unimplemented.
I need to include some kind of zoom for this window.

cuistax, 2018-06-14:
Ok I understand the problem now, I will look into it.

By the way, may I suggest to add a legend to that useful window, so that we can understand what the different colors mean ? 

That would be extra nice !

Another suggestion I would make is to have a tooltip label that shows up as you move the mouse cursor over the different regions, telling what's inside the region. That would make that window even more  useful.

cuistax, 2018-06-14:
Hi Lulu04,

The device units have been fixed here : #57, you can check it out.

Thanks for reporting this, and let us know if you spot any other issue.

t-edson, 2018-06-14:
> By the way, may I suggest to add a legend to that nice window, so that we can understand what the different colors mean ?

It was planned having a legend. No time to implement before. I will check it now.

Lulu04, 2018-06-15:
Hi Cuistax and Edson, Thanks to the fixes.
But now, I can't compile the device unit: the message 'Adress exceed limits of this device'.
I use the win64 version of PicPas 0.8.7

cuistax, 2018-06-15:
Could you share your program ? Or, better, the smallest program that reproduces the issue ?
 
I tried the following (very simplistic) program on my side (a blinkled), it does not produce any error:
```pascal
{Sample program to blink a Led on PORTA.RA2}
program BlinkLed;
uses PIC16F1825;

begin
  TRISA := 0;   //all outputs
  while true do
    delay_ms(1000);
    PORTA_RA2 := not PORTA_RA2;
  end;
end.
```

t-edson, 2018-06-15:
> But now, I can't compile the device unit: the message 'Adress exceed limits of this device'.
This is a bug found in some , han have two parts:
* Bad error location: FIXED. The error must appear in variabel declaration. Please update.
* Bad hardware limits defined in the compiler.

PIC17 compiler assumed the max Bank number for Enhanced Mid-range devices is 32 so address like $1E0F, are not allowed. 
Can someone tell me how is the max number of bank for the Enhanced Mid-range family? 32, 64, 128? This is some crazy.

cuistax, 2018-06-15:
As of today, the maximum number of banks in any PIC is 64.

But, you may as well put the limit at 256, to be future proof.

The current list of models using 64 banks are (all in the Enhanced Mid Range):
PIC16F15313 PIC16F15323 PIC16F15324 PIC16F15325 PIC16F15344 PIC16F15345 
PIC16F15354 PIC16F15355 PIC16F15356 PIC16F15375 PIC16F15376 PIC16F15385 
PIC16F15386

PIC16F18424 PIC16F18425 PIC16F18426 PIC16F18444 PIC16F18445 PIC16F18446 
PIC16F18455 PIC16F18456 PIC16F18857 PIC16F18877

PIC16F19155 PIC16F19156 PIC16F19175 PIC16F19176 PIC16F19185 PIC16F19186 
PIC16F19195 PIC16F19196 PIC16F19197

cuistax, 2018-06-15:
If you are to choose one model for testing the Enhanced Mid Range, I would recommend the PIC16F19197.

It's the biggest, craziest, most complex PIC in the range. 
The datasheet alone is 708 pages long !

If you have that one working, other models will too....

t-edson, 2018-06-15:
OK. I prefer to set in 64 by now, because some variables in the compiler are resized to the max number of banks, so using 128 or 256 will waste memory.
The directives $SET_STATE_RAM and $SET_MAPPED_RAM must be updated too, to support bigger adresses. No a big problem.

t-edson, 2018-06-15:
Some advances in automate test. An option has been added to the >Tools menu, to compile all the units in /device10, /device16 and /device17 fodlers.

cuistax, 2018-06-16:
Strangely I get display issues on Mac:

[image]

Also, some menus don't display correctly anymore, such as the compiler selection:

PicPas 0.8.7:
[image]

PicPas 0.8.8:
[image]

Could it be that some source files were not uploaded to Github ?

cuistax, 2018-06-20:
Hi,

The Device Units have been updated to take advantage of the new syntax.
You can check them out here :  #58 

Cheers,

closed 2019-07-01 by t-edson