t-edson/PicPas #64: Don't know if this is an issue...or me missing something.

jtannenba, 2023-12-19:
Altered the Hello World blink program for a pic16f886.  When loaded with the pickit3, there is a warning that there are no configuration words in the hex file.  Also program does not work.
[Program.txt](https://github.com/t-edson/PicPas/files/13720966/Program.txt)

This is a copy of the program.  I probably missed something easy.  I used to use Turbo Pascal for work around 25 yrs ago and
trying to come up to speed on microprocessors.

t-edson, 2023-12-21:
Hi. 
I'm afraid I haven't tested PicPas with PicKit. 
I don't touch the compiler for a long time, but I remember the hex file includes the Config Word if you use the directive $CONFIG:

```
{$CONFIG $3FFD}

{$define _CP_ON       =     0x000F}
{$define _CP_OFF      =     0x3FFF}
{$define _WDT_OFF     =     0x3FFB}
{$define _LP_OSC      =     0x3FFC}
{$define _XT_OSC      =     $3FFD}

{$CONFIG _CP_OFF, _XT_OSC, _WDT_OFF }

{$CONFIG _CP_OFF _XT_OSC _WDT_OFF }
```

Updating the PicPas compiler is in my TODO list.

jtannenba, 2023-12-21:
I'll try that tomorrow.  Thank you.

jtannenba, 2023-12-22:
I added {$CONFIG $00EA}  which should be HS, wdt on, Boren off and copy protect off.  On PicKit3, should be Config1 00EA and Config2 0700. Not sure about the second word which is the write protect.  The program does not work either. The debug show it going, but the chip isn't going.
I didn't do any $define statements, just the config statement.  How do we define config reg2 (write protect bits)
If I seem dense, please forgive me.  My brain is 80 yrs old and not as functional as it was.

t-edson, 2024-02-04:
Hi. 

> Edit: Don't have a PIC16F84, so tried a PIC16F628A which also has a 13 bit config and it works ok. Is there a Syntax guide? 
There is a user Manual and  Technical Documentation in /Docs folder, only in Spanish by now. I think it should be easily translated with a translator online.
> And, should it work on a PIC16F886? 
It should. But maybe you can use some other PIC programmer hardware or some way to set the second config word.
> How do you add the 2nd Config word?
Second config word is not implemented in the current version. I will put this requirement in the TODO list.

closed 2024-02-05 by jtannenba